shamansystems/Cheap-FOCer-2 #3: B6B-PH-K-S(LF)(SN) Discontinued

Helno, 2020-05-01:
Discontinued. 

https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_JST-Sales-America_B6B-PH-K-S-LF-SN_JST-Sales-America-B6B-PH-K-S-LF-SN_C160289.html

This appears to be a compatible replacement. 

https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_JST-Sales-America-B6B-PH-K-LF-SN_C274409.html

bj97301, 2020-05-10:
Good find.  @shamansystems if it looks good to you I will replace it in the Lcsc bom file.

shamansystems, 2020-05-12:
https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_BOOMELE-Boom-Precision-Elec-PH-7A_C10398.html

https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_CJT-Changjiang-Connectors-A2001WV-7P_C225193.html

These look compatible as well. Price points cater to different quantities though. I'm down with replacing with helno's suggestion, either of mine, or all of them since we have an "alternatives" list going.

Helno, 2020-05-12:
Those two you listed are 7 pin. 

I'm sure there is something out there that is cheap and available in multiples of 5.

shamansystems, 2020-05-13:
Ah yes my mistake. I’ll look again

Helno, 2020-05-19:
How about these. 

https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_CJT-Changjiang-Connectors-A2001WV-6P_C225192.html

Minimum order 10 cost for 10 is $0.37

Or these. 

https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_BOOMELE-Boom-Precision-Elec-PH-6AK_C146086.html

Minimum order 20 cost for 20 is $0.35

The boom ones are cheaper but the BOM will need a note for people ordering more than 5 at a time so they don't order 100 of these when they don't need to.

shamansystems, 2020-05-21:
I'm down for the cheapest approach even though it's more connectors than needed. Notes can be added to alert people of the MOQ

Helno, 2020-05-21:
Even with the increased min it lowers the over all cost. 

What is the easiest way to update that spreadsheet? I don't see an easy way to do a pull request on it.

closed 2020-05-24 by Helno