shamansystems/Cheap-FOCer-2 #4: U1/AMS1117-3.3 and other zero rotation error in JLCPCB

hackerboygn, 2020-05-13:
check U1 and other  other multi-pin components in https://github.com/shamansystems/Cheap-FOCer-2/blob/master/guides/ordering/images/a33759f1b319649d610ba648f7687170_html_a3977b4e.png, as you see , their direction error.

This footprint Package_TO_SOT_SMD:SOT-223-3_TabPin2 and other in KiCAD library conforms to the IPC standard, but JLCPCB NOT!
JLCPCB stipulates that the positioning hole of the package materials tape and reelis on the left side is zero rotation, more information ref http://club.szlcsc.com/article/details_116_1.html, http://club.szlcsc.com/article/details_1569_1.html and https://www.maximintegrated.com/en/design/packaging/package-information/smd-tape-reel-info.html.

Although reminded everyone in the  ordering guide, but I think should be modified use the JLCPCB standard librarys.

Thanks share this project again!

shamansystems, 2020-05-13:
Hey thanks for your input!

Yes true that JLCPCB doesn’t use KiCADs placement standard (IPC) and I was aware of this. However, JLCPCB does a good job correcting the placement of components in their DFM review. I’d rather not deviate form the IPC standard in case people want to use other vendors. As long as it works the way it is, I’d rather not fix what ain’t broken.

closed 2020-05-21 by shamansystems

hackerboygn, 2020-05-21:
> Hey thanks for your input!
>
> Yes true that JLCPCB doesn’t use KiCADs placement standard (IPC) and I was aware of this. However, JLCPCB does a good job correcting the placement of components in their [...]

I think you are right, I have made one too, thanks again！