shamansystems/Cheap-FOCer-2 #7: nRF51 using Windows

MegaC-C, 2020-05-18:
The vesc-tool SWD programmer couldn't detect my Waveshare nRF51822 module.
So i used openocd (for Windows) and the ST-Link V2 to flash the nRF51822 chip. I didn't know openocd and still don't undestand it, but here is how it worked for me:

-download & extract the precompiled openocd for Windows from here: https://gnutoolchains.com/arm-eabi/openocd/

-open the bin folder (it should contain two .dll files and the openocd.exe)
[image]

-download the openocd.cfg file from here: https://github.com/vedderb/nrf51_vesc

-download the correct .bin file for your module from here: https://github.com/vedderb/nrf51_vesc/tree/master/build_all
(my module has 32kB RAM and 16MHz clock, so i chose nrf51_vesc_ble_32k_16m_rx11_tx9_led3.bin)

-add the two files to the bin folder of openocd
[image]

-connect the nRF51 module (SCLK, SDO, VDD, GND) via the ST-Link v2 as described here: https://github.com/vedderb/nrf51_vesc/blob/master/README.md

-open command prompt (windows + cmd) and type:
**cd C:\ _yourPath_ \openocd-20200503\OpenOCD-20200503-0.10.0\bin**        (now you are in the openocd bin folder)

-then type:
**openocd -f openocd.cfg -c "init" -c "halt" -c "nrf51 mass_erase" -c "program nrf51_vesc_ble_32k_16m_rx11_tx9_led3.bin verify reset exit"**        (if you used another .bin file, then type the name of that .bin file of course)
[image]

-wait until flashing is finished and connect the vesc RX TX pins to the correct pins of your nRF51 module (pins RX<>p0.09 and TX<>p0.11 in this case)

-power the vesc up and connect BLE via the android app
 
done

MegaC-C, 2020-05-18:
oh and i soldered a 10uF cap between vcc and gnd of the nRF51 module just as Mr. Vedder suggested, i don't know if it was necessary

shamansystems, 2020-05-19:
So what was the issue and what does it have to do with the CFOC2?

MegaC-C, 2020-05-19:
My issue: the CFOC2 could not act as SWD programmer for the nRF51822 module I have (no device detected).

You are right, it is not specifically an issue of the CFOC2. But maybe other CFOC2 users might run into the same problem and it took me quite some time to find a solution. So I thought it helps when all the info is in one repo.

wrong place? ( sorry i'm new to github)

shamansystems, 2020-05-21:
It's fine. I'll close the issue but the info will be retained on this repo.

closed 2020-05-21 by shamansystems