shamansystems/Cheap-FOCer-2 #15: conf_general.h refers to hw_FOCer.c while ref to hw_FOCer_2.c commented out

shanghaijo, 2020-06-09:
Line 153: #define HW_SOURCE "hw_FOCer.c"

while line 150 is commented out: //#define HW_SOURCE "hw_Cheap_FOCer_2.c"

https://github.com/shamansystems/Cheap-FOCer-2/blob/67b7fbcc121aee1f70713e3b77ed3238eb31ad52/firmware/Source%20Code/bldc-master%205.01/bldc-master/conf_general.h#L153

lachlanh, 2020-07-26:
Made the above change and built the firmware. Builds and uploads ok, but any app other than UART, will work initially after flashing the firmware and then cause all 3 leds to flash on subsequent reboots of the controller at a regular interval and I am not able to connect to the controller.
Is the source in the repo correct ? 
Is their a fork of the bldc repo that is more up to date ?
(I am trying to build a custom app so prebuilt firmwares are working fine, but I need one I can build, also awesome project, loving the hardware!)

haidongt, 2020-09-05:
I ran into the same issue. I'm also developing a custom app and the issue is only observed when the firmware is built from source. After some debugging I found that the issue is only present when rebooting after a mcconf or appconf save. If I don't write to flash at all and reboot the board works perfectly fine. I think it must be something with the flash memory but I haven't figured out.

lachlanh, 2020-09-05:
Yup that is the same that I have observed, any change to settings and it fails on the reboot. Have worked around it by building from source with my appconf and mcconf headers compiled in, solves the problem for the present. Writing flash does seem to be a likely cause, was wondering if it was something to do with how I flashed the firmware and the bootloader initially but haven't made any progress figuring it out. If you do figure it out, please post again.

nhedlun1, 2021-01-20:
Hello, im facing the same issue. Did you ever find the source of why this happend? I also tried to just upload the Bootloader but it fails to boot then also. Could there be something with the bootloader? The pre-built firmware in the "latest" folder works as expected though.

nhedlun1, 2021-01-21:
Ok i think this is a compiler issue. I solved the problem by installing GNU Arm Embedded Toolchain: 7-2018-q2-update June 27, 2018, the newest version of arm-none.eabi-gcc does not work. After a APP setting is changed it goes into a boot loop. 
I followed these steps, hopefully someone else is helped by this.

Issue on Vedders github describing almost same behaviour: https://github.com/vedderb/bldc/issues/210

How to install specific toolchain: https://freeelectron.ro/arm-cross-compiler-tutorial-stm32/

Link to toolchain i used: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads/7-2018-q2-update

After i installed the toolchain as per the instructions, i had to run "source ~/.bashrc" or restart terminal to update file. 

Hope it helps someone!

haidongt, 2021-01-24:
That also solved my issue. Thanks a lot, nhedlun1!

lachlanh, 2021-03-02:
Thanks nhedlun1, that fixed the issue for me also. Seems to need that version of the arm gcc that you mentioned to be able to write to flash (I was using the version from the ubuntu package repos before that).