shamansystems/Cheap-FOCer-2 #17: I was told by JLCPCB that they will no longer order more DRV8301 and found some potential problems with DRV8301

victordlee, 2020-07-02:
I tried to put in my order for FOCer-2 and noticed the DRV8301 was out of stock again.  So, I asked JLCPCB when they would get more in stock.  They told me that they do not plan to order more DRV8301 in the near future due to low demand.  In any case, I asked the question through their chat line so if you want to drive up demand, I suggest you pick chat a couple of times to generate more demand for the part.

I also notice this potential problem with DRV8301...

https://e2e.ti.com/support/motor-drivers/f/38/t/668809

DrPii, 2020-07-06:
Hey,
just wanted to comment on this.
Any plans of integrating the new chip mentioned in the post above?

> The DRV8320R is a newer device with improved features and a more thorough test program

thanks and great Job so far :)

shamansystems, 2020-07-06:
@victordlee unfortunately I don’t have control over what JLCPCB has in stock. As of now, It doesn’t seem they have the DRV8320 in stock either. While I know it’s an older device, the DRV8301 is what has been stocked so far and can be handled by their SMT service. The DRV8320 is a QFN package and difficult for typical DIYers to handle and solder.

DrPii, 2020-07-06:
Ah thanks for that.
I didn't even check if they had the 8320 in stock. 

Keep up the great work 😁👍

transcendReality, 2020-07-08:
This guide no longer works. They simply don't have the components. Out of the entire bom and cpl I ordered from LCSC, they only delivered the 100v radial capacitors, a momentary switch, and the IMU. Not a single ceramic cap- no USB connector, no connectors period. I don't have the immense level of patience it would take to figure out what I'm missing, and feel like I've blown my money.

This certainly wasn't as easy as just following a guide.

I don't understand why you just didn't offer people kits?

shamansystems, 2020-07-09:
@transcendReality You're more than welcome to bring your issue and it's details to the support thread mentioned in the readme of this repo. I don't offer kits because this project is meant to enable DIYers and for others to create kits if they want. There's already someone making kits for Europe.

closed 2020-07-09 by shamansystems