shamansystems/Cheap-FOCer-2 #23: BMI160 Out of Stock at https://lcsc.com/

unRARed, 2020-12-13:
Also, out of stock at Mouser and Digikey. Will update if I find somewhere where it is available.

Thunkar, 2021-01-15:
Following, really bad news for a DIY OneWheel...

Thunkar, 2021-01-15:
The only viable alternative I've found is to order a complete board such as this one:

https://www.amazon.es/HiLetgo-Palstance-Accelerometer-Gyroscope-Acceleration/dp/B082M8PTK7/ref=sr_1_4?dchild=1&keywords=bmi160&qid=1610728010&sr=8-4

...and remove the chip using the reflow station. Digikey shows that availability might be limited at least until June, so I might just do that.

unRARed, 2021-01-15:
I was able to purchase a pack of 5 from an eBay listing... from China. Took
about 2 weeks to arrive. Haven’t built anything up yet though.

Mitchlol, 2021-01-22:
the lsm6ds3 can be placed in the same spot, it would require custom firmware for now (easy small change), however there is talk to make the firmware auto detect the populated IMU.

closed 2021-04-28 by shamansystems