shamansystems/Cheap-FOCer-2 #24: 63ZLJ330M10X20 Out of Stock at https://lcsc.com/

unRARed, 2020-12-13:
[image]

In stock [at Mouser](https://www.mouser.com/ProductDetail/Rubycon/63ZLJ330M10X20?qs=%2Fha2pyFadug%2FAvrZoUF8PMlUvvJxatQqQWG%2FmtfYZYOzNEOAZQquww%3D%3D), however.

NickLD, 2020-12-14:
@unRARed I noticed those same caps were out of stock when I ordered parts for a batch of 5 a few weeks ago.
I used the part finder in LCSC to find a drop-in replacement. I used the [C357629](https://lcsc.com/product-detail/Aluminum-Electrolytic-Capacitors-Leaded_BERYL-Electronic-Tech-RG063M331LO10-20TA-1A1E_C357629.html) which have worked just fine on the boards I've made. Only spec difference is they are rated for 7k vs 10k Hrs, but they should be just fine.

unRARed, 2020-12-15:
Thanks, I'll grab these in the case the backorder is indefinite. =)

nespotts, 2020-12-30:
I also just found that there is the same capacitor in just a different length cylinder available (10x25mm) [63ZLJ330M10X25](https://lcsc.com/product-detail/Aluminum-Electrolytic-Capacitors-Leaded_Rubycon-63ZLJ330M10X25_C694362.html).

unRARed, 2020-12-30:
I ended up going with Nick's option. Just received them yesterday so haven't built anything up yet. @shamansystems should I keep this open?

NickLD, 2020-12-30:
Just created a quick PR to add the two alternative options discussed in this issue to the 'out-of-stock' page.
Meant to do that weeks ago, my bad ;)

unRARed, 2020-12-31:
Awesome!

closed 2020-12-31 by unRARed