shamansystems/Cheap-FOCer-2 #33: STM32F405RGT6/7 alternatives.

BigHuz, 2021-04-22:
Hello everyone so I have done a little research for ST microcontroller on cf2. I was  thinking to myself is it possible that STM32F407RGTxx would be a sutible alternative for microcontroller that is listed in cf2 bom. Also there is a STs paragraph that says that STM32F405xxxx and STM32F415xxxx are used in the same fields and have same specs. So STM32F407xxxx and STM32F405xxxx microcontrollers are from the same microcontroller lineup and STM32F415xxxx has same specs as listet ST microcontroller for cf2. Now which one of this above mentioned chips would you recommend for alternative chip and if possible and safe to use some copy cat chip form other companys (something like Giga Devices)?? Sorry if I have repeated myself I am very new to world of microcontrollers. Thanks in advance,
BigHuz

shamansystems, 2021-04-28:
The VESC project only supports one kind of MCU at the moment. This means the CFOC2 will only support this same MCU. No other substitutes are compatible.

BigHuz, 2021-04-28:
Not even STM32F305RGT7 the one with bigger working temperature range. Is so thank you very much.

shamansystems, 2021-04-28:
If the only difference is the operating temperature range, then I suppose it would work.

closed 2021-04-28 by shamansystems