shamansystems/Cheap-FOCer-2 #34: 200A / 12 FET version.

xromansx, 2021-04-27:
Hi,
 I was interested if there are any plans to build 200A version, 200 Amps means it will need 12 FETS and better cooling or everything is much more complicated ? I'm new to all this. Thanks

bj97301, 2021-04-27:
Yep. Feel free to checkout shaman's posts at the forum below:

https://forum.esk8.news/t/little-focer-84v-5kw-vesc-based-controller/50110/256

xromansx, 2021-04-27:
I saw it but it's only 75Amps and as I know that minimum for couple of month you wouldn't buy it OOS ... that's why I was thinking what can be done to Cheap FOCer to increase the amps ... or it there will be high A version I need it for 12S. Intend to use it with efoil.

bj97301, 2021-04-27:
[image]

xromansx, 2021-04-27:
I saw this prototipes before they look nice but as I know they where never realized.

bj97301, 2021-04-28:
Not yet. Little focer is being released slowly so soon enough.

xromansx, 2021-04-28:
Do you know if there are any schematics with 12 FETs interesting to see what FETs are used and how different is the scheme?

bj97301, 2021-04-28:
Probably not much. He's going more retail focused with the rest of the controllers.

shamansystems, 2021-04-28:
There aren't any existing schematics with 12 FETs or anything more powerful than the CFOC2 for open hardware. I do have plans for a more powerful CFOC3 in the future but that has not yet been started. You are more than welcome to alter the existing open source and available KICAD project files and create a 12 FET version for yourself.

xromansx, 2021-04-28:
Thanks for all your efort and time you are putiong in this project. Is your comercial version based on DRV8353RS  ? And CSD18536KCS and NCEP60T20 in their datasheet it tells that they can support 200A but in real life more then 50A geting them to hot is something else is limiting max Amps or with good cooling they can support much more? Regarding 12 FET I wanted to understand they are used in paralel to decrease RDS and to redistribute the heat. 
Thanks

shamansystems, 2021-04-28:
The Little FOCer does use the  DRV8353RS and operates up to 84V with no problem.

The datasheet numbers for max current on MOSFETs is a very misleading number. Real life performance is much different and depends on system cooling.

Yes parallel MOSFETs reduce the total Rds(on) and therefore increase the total current that can be delivered.

xromansx, 2021-04-28:
Thanks for your reply, will be waining for your 12 FET version in meantime will try to experement with your cheap FOCer 2 maybe increasin nr of fets to see if can get to higher currents, this problem with parts shortage is also anoing.
Thanks.

shamansystems, 2021-04-28:
To be clear, I never said I was working on a 12 FET version. But indeed a more powerful version. More details on this after I have prototypes in the future.

closed 2021-04-28 by shamansystems