shamansystems/Cheap-FOCer-2 #40: Replacement of parts

snapo, 2022-07-04:
Heya.... 
First of all Thank you sooo much for all that work you have done.
i did want to order 5 and expected a 120$ price tag.... 
and got a little surprised. My Bill got to 300.50 USD .....

The most expensive thing is the 
DRV8301DCAR , this costs 32.4 USD a piece because of the current global shortage. This price isnt worth the silicone inside it.
the cheaper chinese alternative is the DRV8301DCA (same pinout size, but 4 have to get a new trace) and costs only 7USD
Another alternative is is a different driver all together instead of TI.

Another way to make it way cheaper is replace the expensive STM32 (11USD piece) with a raspberry pi pico ... both of them run at 168Mhz but the pico chip is 80% less expensive :-) jlpcb also has it in their smt service.

Best regards keep up the good work

Koxx3, 2022-07-04:
running vesc on rpi ?! good luck ! 😅

shamansystems, 2022-07-05:
Hey there. I'm working on the CFOC3 right now to address the current shortages and also improve performance. No more DRV. Still has the STM32 since the VESC firmware only runs on that.