shamansystems/Cheap-FOCer-2 #45: NOOB question: I cant find any 2W version of R57-R61 on LCSC/JLCPCB.

Meserlion, 2022-12-17:
Can i use 1W instead? Or can i use 2X1w to get the same? This is the first time im ordering any kind of PCB. Your guide is amazing BTW.

Meserlion, 2022-12-17:
C2931966, This is the only 3W i can find. Will that work?

charclo, 2022-12-17:
That will probably work. It should be a 2512 resistor with 1mOhm and at least 2W.
Make sure to order the 1.0 version of the cheap focer 2, it is in the developer branch. Also, Shaman is working on a cheap focer 3 with some improvements. If you don’t need it soon, you can wait for that

Meserlion, 2022-12-17:
So... what happens if i already ordered 0.9?

charclo, 2022-12-17:
Not that much, there were some small improvements. Just make sure to use the firmware for 0.9 version. It is included in the latest vesc tool, or the firmware included in the github repo of the cheap focer 2

Meserlion, 2022-12-17:
Thanks, i think ill use 0.9 for making all my mistakes on. Then i can upgrade to v3 when that is ready.

charclo, 2022-12-17:
👍

charclo, 2022-12-17:
👍

Meserlion, 2023-01-27:
Its alive! I connected to VESC tool and MCU is responding fine.